CR-Formula/STM32-Examples #11: Shift Light Circuit Design

APBashara, 2023-06-24:


APBashara, 2023-09-21:
Requirements for LED Driver
PWM input 
Runs off of 5V or 3.3V
Runs 6-10 LEDs --> preferably 8

Will also need to spec LEDs
LEDs need to be bright enough to be seen in sunlight

CalebRezmer, 2023-09-28:
https://docs.google.com/spreadsheets/d/1XcHvZkbIPLKPrKGtF2rANDc3LexFMXuKf3GEuLuzu4w/edit?usp=sharing

APBashara, 2023-10-07:
[TI 16 Channel option](https://www.ti.com/product/TLC5940)
[Sparkfun Resources](https://www.sparkfun.com/products/10616)
[Instructables Demo](https://www.instructables.com/Arduino-and-the-TLC5940-PWM-LED-Driver-IC/)

APBashara, 2023-10-07:
I think that we should use the TI IC as it has both Digital and Serial control as well as multiple reference designs.

APBashara, 2023-10-07:
For LED colors, generally most systems are designed to go Green -> Red -> Blue
Will have 6 Green -> 2 Red -> 2 Blue, or at least that ratio

APBashara, 2023-11-01:
Few notes on dimensions and design
Need to move the LEDs to the top of the board
Try and shift dimensions to 5.38" wide

closed 2024-06-15 by APBashara