CR-Formula/STM32-Examples #12: Thermocouples

APBashara, 2023-06-24:


APBashara, 2023-06-24:
Controls wants 3 Thermocouple inputs so add 5 at least

clairejhaas, 2023-08-10:
[K Type Analog Output Amps](https://www.adafruit.com/product/1778)

Are we also speccing the thermocouples too or just the amps? @APBashara

APBashara, 2023-08-10:
Should be just the Amps. I will double check with Logan. I am pretty sure the Thermocouples they found are built into the brake pads.

APBashara, 2023-08-11:
All of the Thermocouples will be K type. Controls has yet to pick out the exact ones.

APBashara, 2023-08-29:
Look into how we can connect Thermocouples
Need to verify that we can solder them into our connectors or consider making a daughter board to add them to our main harness

APBashara, 2023-08-31:
Looks like the thermocouple connectors are generally recommended as they are specific metals; however, we can solder them. We can look at buying specific wires or getting PCB header connectors for a daughter board. We just need to make sure we have a good plan for mounting any daughter bords we make.

APBashara, 2023-09-01:
https://digilent.com/shop/thermocouple-wire-k-type/

APBashara, 2023-09-18:
Controls has said that they will spec the K type thermocouples as they have to fit in the brake pads correctly. They will have standard two prong connectors on them. We will have to convert from that to an analog out signal using thermocouple amps. They said that they only need two.

closed 2023-09-18 by APBashara