CR-Formula/STM32-Examples #2: Spec IMU

APBashara, 2023-06-24:


APBashara, 2023-06-24:
https://www.omniinstruments.co.uk/lpms-curs-9-axis-imu-ahrs-motion-sensor-oem-version.html

vaughn-miller, 2023-06-27:
https://www.digikey.com/en/products/detail/tdk-invensense/IAM-20680HT/15792840 good automotive solution but requires some more pcb work

vaughn-miller, 2023-06-28:
https://docs.google.com/spreadsheets/d/1dcyz6iV356iEcNz_fcOY-OLWTQ99qWUj6Q1IHMWERbc/edit?usp=sharing

APBashara, 2023-06-30:
Ok I like the sparkfun one haveing the best general specs, only concern is hooking it up
We had lots of issues with I2C on 27 so I don't want to rely on that but if we could use SPI that would be a good option
I think we should plan on testing the Sparkfun IMU we have with SPI on 27 and see if it works well, if so I think we do that otherwise we can go with the alternative $170 one with CAN or RS232 capabilities. What do you think about this @vaughn-miller?

vaughn-miller, 2023-06-30:
Works for me, I can try to get it going if we have dev boards in ames in the next couple weeks

APBashara, 2023-06-30:
Yeah, it will just be using the system on 27 to test. Would like to try to remove the existing I2C lines and add ones just for the IMU first; the downside is the car has to be running to test this. The SPI test may require more work and possibly soldering wires to the back of the mainboard to test things. Will most likely be very messy but that's ok cause it is temporary. The code is something I should be able to do rather quickly. Let me know if its something you want to mess with.

vaughn-miller, 2023-07-03:
Why does the car have to run for i2c?

APBashara, 2023-07-04:
When the engine runs, I2C stops working. I'm pretty sure its due to noise issues so if we are planning on using it for 28 then we need to have it 100% functional on 27 first.

closed 2023-07-13 by APBashara