CR-Formula/STM32-Examples #3: MCU Board Design Rev 1

APBashara, 2023-06-24:
Begin designing a basic MCU board
Breakout pins
Add programming headers
Super basic revision 1

APBashara, 2023-06-24:
https://docs.arduino.cc/resources/schematics/ABX00063-schematics.pdf

clairejhaas, 2023-06-24:
@MHurley117 I created a footprint in Altium using the component wizard for the microcontroller LFQP144 package (datasheet pg 263). Please review. Symbol currently has no pin names until we create the pin outs

APBashara, 2023-06-26:
I want to make sure that we are keeping analog and digital signals as far apart as possible. I read a few articles that talked about a Microcontroller board design that was 4 layers and was Digital - Power - Core - Ground - Analog. I think that this is a good idea and that we should try and implement this. We can add more digital or analog layers if need be on the top/bottom.

APBashara, 2023-06-30:
Total Sensors List
CAN Bus to ECU
Extra CAN Bus -- may need to use this to communicate with the PDM for logging
Analog -- 4x suspension pots, 2x brake pressure, 1x steering, 4x Thermocouples for brakes
SPI -- 1x IMU
I2C -- 1x As an extra
Digital -- will want to break out at least 4 interrupt pins and some GPIO pins as well, mostly for "Just in case"
UART -- 2x at least, one for Nextion and at least one extra -- will be able to use as LIN bus if need be that way

***These are subject to change slightly but will give us an amazing start for Rev 1

clairejhaas, 2023-07-03:
Pins named on MCU footprint and symbol. Please notify if remapping pins so I can adjust footprint. Working on schematic and choosing board connectors. Posted important page numbers in MCU datasheet

- Page 59: LQFP 144 Pin-Out
- Page 204: PCB schematic
- Page 264: MCU footprint info

clairejhaas, 2023-07-08:
We will be using 0603 (1608 metric) for capacitors and resistors as that is the standard for most board assemblies. We should ask the sponsor if they will provide passive components, as many have bulk stock of standard 0.1uf capacitors and various resistors.

clairejhaas, 2023-07-18:
@MHurley117 Choose a voltage regulator and look at SMPS/LDO mapping

Component selection (minus LDO) is chosen for SMPS as well as capacitors, resistors, and inductors needed for power supply. I drew traces on VDD/VSS pairs, VCAP, VBAT. 

- [ ] SMPS Traces
- [ ] VDDLDO Traces
- [ ] VDD USB Configurations and Traces

clairejhaas, 2023-07-23:
SMPS and LDO power supply scheme is configured. 

- [ ] Configure USB - will use USB-B. 
- [ ] Clock oscillator

Need the Pinout as soon as possible to do traces to connector

APBashara, 2023-07-24:
Is there a reason that you are choosing USB-B @clairejhaas?

clairejhaas, 2023-07-24:
[Altium USB-C) (https://www.youtube.com/watch?v=t67KcQHhIC4)
[20 pin JTAG stuff](https://www.st.com/resource/en/user_manual/um1075-stlinkv2-incircuit-debuggerprogrammer-for-stm8-and-stm32-stmicroelectronics.pdf)

clairejhaas, 2023-07-28:
[Pushbotton Switch](https://www.digikey.com/en/products/detail/1825910-6/450-1650-ND/1632536?curr=usd&utm_campaign=buynow&utm_medium=aggregator&utm_source=octopart)

clairejhaas, 2023-07-29:
Power config is complete, will work on signal traces to Deutsch connectors next

clairejhaas, 2023-08-10:
Finished routing the board. @MHurley117 Given a Signal-GND-PWR-Signal, do you use the default via setting or change everything to thru vias? The default via setting creates two vias, usually a blind and a skip via.

APBashara, 2023-08-10:
Board design looks great, I have a few things that I would comment on and want @clairejhaas's and @MHurley117's thoughts on

Instead of a switch for the power input, I'd think we should use a jumper instead, less chance of it getting knocked around

Could we maybe move the Power in, I2C, and CAN signals away from the ADC connections? Differential Signals such as CAN and I2C can cause lots of interference with Analog inputs. The Power in is less of an issue but I would like to try and make the most of analog signal integrity if at all possible. 

Finally, at Comp the design judge mentioned not having any right angles, I am unable to find any so good job there. The other thing that was mentioned was having traces come straight out of the pads for things like resistors. I see a few that have traces coming out at 90deg instead of 180. Is this something that we need to address?

Thanks for the great work @clairejhaas and @MHurley117! I can't wait to see this thing.

clairejhaas, 2023-08-10:
I am not sure about the 90 degree thing, but I am sure @MHurley117 can answer this. If we have an extra 8 dollars/want to buy the D connector, I can move the digital signals

clairejhaas, 2023-08-12:
@APBashara In case you didn't see in Electrical, PCB is ready to order, you can go in and double check connections if you want. I am going on vacation, so talk to Michael about getting it ready for manufacturing/assembly.

APBashara, 2023-08-21:
I think that the PCB looks great. Great job on getting it done over the summer. The only thing that it is missing is the LoRa WAN chip. Do we think that Rev 1 should forgo having wireless for the sake of testing earlier? Or should we focus hard on getting the wireless figured out this week?

clairejhaas, 2023-08-21:
Do we have a particular LoRa WAN chip that @MHurley117 wants us to use? I personally say forgo it to test earlier, I can do by general if we choose the chip by midday Wednesday

APBashara, 2023-08-23:
We don't have a specific one picked out yet. That was going to be goal #1 for workday this week. We can talk about it during the meeting this week.

closed 2023-08-26 by APBashara

clairejhaas, 2023-09-07:
- [ ] ADC
- [ ] rpack
- [ ] LoRaWan

clairejhaas, 2023-10-12:
[CycloneRacing.zip](https://github.com/Cyclone-Racing/CR28_Telem/files/12883204/CycloneRacing.zip)

1 oz copper. Stencil is a GTP file per requested