CR-Formula/STM32-Examples #9: Spec Board Connectors

APBashara, 2023-06-24:
Need to find board to wire connectors for our main harness

APBashara, 2023-06-26:
Possibility for TE Autosport connectors
I think we should try and stick with TE stuff to keep it consistent across the car

APBashara, 2023-06-30:
Really nice expensive ones Chris is trying to get sponsored
https://www.te.com/usa-en/industries/autosport/as-connectors-ranges.html

Other Solid option
https://www.te.com/usa-en/product-CAT-D799-H342.html

General Wire to Board list from TE
https://www.te.com/usa-en/plp/wire-to-board-connectors/Y30o9.html

clairejhaas, 2023-07-07:
Here is a connector that provides right angle, non-panel connection at around $23:

[https://www.te.com/usa-en/product-344108-1.html](https://www.te.com/usa-en/product-344108-1.html)

The associated housing ($8) and necessary pins ($0.40 each): 
[https://www.te.com/usa-en/product-344111-1.html](https://www.te.com/usa-en/product-344111-1.html)
[https://www.te.com/usa-en/product-344113-1.html](https://www.te.com/usa-en/product-344113-1.html)

clairejhaas, 2023-07-19:
$5 [https://www.te.com/usa-en/product-DT06-12SC-P012.html?compatible=DT13-48PABCD-R015](https://www.te.com/usa-en/product-DT06-12SC-P012.html?compatible=DT13-48PABCD-R015) 
$35 [https://www.te.com/usa-en/product-Dt13-48PABCD-R015.html](https://www.te.com/usa-en/product-Dt13-48PABCD-R015.html)
$4 [https://www.te.com/usa-en/product-DT06-12SA-P012.html?compatible=DT13-48PABCD-R015](https://www.te.com/usa-en/product-DT06-12SA-P012.html?compatible=DT13-48PABCD-R015)
$4 [https://www.te.com/usa-en/product-DT06-12SB-P012.html?compatible=DT13-48PABCD-R015](https://www.te.com/usa-en/product-DT06-12SB-P012.html?compatible=DT13-48PABCD-R015)

APBashara, 2023-07-19:
Data sheet for board to wire
https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Specification+Or+Standard%7F114-151046%7FA%7Fpdf%7FEnglish%7FENG_SS_114-151046_A.pdf%7FDT13-48PABCD-R015

APBashara, 2023-07-20:
Total Pin Count as of Right Now
CAN -- 4
Suspension Pots -- 4
Brake Pressure -- 2 
Brake Temp -- 8
IMU -- Max 3
GPS -- Max 3

24 pins max

APBashara, 2023-07-20:
Having spent some time thinking about what pins we will need and exploring TE's options, I think that the above links in [this message](https://github.com/Cyclone-Racing/CR28_Telem/issues/9#issuecomment-1642883196) is the best option. It will give us lots of flexibility and allow us to have keyed connectors so no one accidentally plugs in the wrong connector. My only concern is the size of it relative to the PCB. Do we have any estimates as to how big we think our PCB will be?

clairejhaas, 2023-07-20:
The PCB will for sure have to be 5.75" long assuming we choose this connector. I am not entirely sure, but I plan on making the microcontroller more of a rectangle if we go this route, trying to keep it under 3.5" wide