CR-Formula/STM32-Examples #16: Order PCBs

APBashara, 2023-08-21:


APBashara, 2023-08-31:
Waiting on sponsor info
https://www.pcbway.com/project/sponsor/Cyclone_Racing_CR_28_071546a3.html

APBashara, 2023-08-31:
PCB Way gave us $20 for our first order

APBashara, 2023-08-31:
Just PCB -- $80.86
Stencil -- $10
Shipping -- $25.90
Total -- $116.76

Total with Full Assembly -- $163.76

APBashara, 2023-08-31:
I think that the best option is to just buy a stencil and do assembly ourselves with the 1316 Coover oven

APBashara, 2023-09-01:
Had multiple Drill files in the zip, corrected and waiting on approval from PCBWay

Jade said that she is donating money and would help cover the cost for PCBs

APBashara, 2023-09-02:
A Few issues with the board Design, Closing until they are fixed

clairejhaas, 2023-09-13:
Created new workspace for this semester. SD card and analog stuff updated this weekend

APBashara, 2023-10-05:
PCBs will be ordered by end of the week
Have a sponsor
Will have to pay for stencils

closed 2023-10-14 by APBashara