CR-Formula/STM32-Examples #22: SD Card Reader

APBashara, 2023-09-02:


clairejhaas, 2023-09-05:
We are doing Micro SD card I'm assuming?

APBashara, 2023-09-06:
Yes, I think that it will be a good thing to have so that we can log locally

clairejhaas, 2023-09-16:
https://www.onsemi.com/pdf/datasheet/bss138w-d.pdf

clairejhaas, 2023-09-18:
@APBashara I am using SPI4 on the STM32 CubeIDE. PE2, PE4, PE5, PE6 for the SD card. The NSS will need to be set, I am not sure if it is input or output, but will be used to CS on the microSD. Please reserve these pins

clairejhaas, 2023-09-21:
[](https://www.ti.com/lit/ds/symlink/txb0104.pdf?HQS=dis-dk-null-digikeymode-dsf-pf-null-wwe&ts=1695249347568&ref_url=https%253A%252F%252Fwww.ti.com%252Fgeneral%252Fdocs%252Fsuppproductinfo.tsp%253FdistId%253D10%2526gotoUrl%253Dhttps%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Ftxb0104)

clairejhaas, 2023-09-28:
https://datasheet.ciiva.com/15174/getdatasheetpartid-716154-15174887.pdf?src-supplier=Verical

closed 2023-09-28 by APBashara