CR-Formula/STM32-Examples #26: SD Card Firmware

APBashara, 2023-09-07:


clairejhaas, 2023-09-28:
[image]

priedieux, 2023-10-03:
> [image]
Claire, what mode did you set Spi1, 3, and 4, USART 1 & 2, and I2C1? thanks

APBashara, 2023-10-03:
We will get the ioc file imported into the main CR_28 project at some point. The mode shouldn't matter, as long as the pins are the same.

priedieux, 2023-10-14:
https://stm32f4-discovery.net/2015/08/hal-library-20-fatfs-for-stm32fxxx/

priedieux, 2023-10-14:
http://stm32f4-discovery.net/2014/07/library-21-read-sd-card-fatfs-stm32f4xx-devices/

priedieux, 2023-10-21:
https://01001000.xyz/2020-08-09-Tutorial-STM32CubeIDE-SD-card/

APBashara, 2023-10-26:
Using FAT Filesystem, figuring out how to interface with that over SPI Bus
Referencing Digikey Videos

priedieux, 2023-11-16:
https://www.keil.com/pack/doc/CMSIS/Driver/html/group__mci__interface__gr.html

CMSIS driver info

closed 2024-06-15 by APBashara